citysu/csiread #2: Known issues

citysu, 2020-10-15:
1. noise will change after calling get_scaled_csi()
2. get_scaled_csi_sm() value error when Ntx is not 1.
3. csiread.CSI may not work well on big-endian computers.
4. in matlab, read_log_file drops the last two packets(Atheros)

serrhini, 2021-01-13:
Hi, tnx you for your efforts, I tried to use this tool but I have an error see 
Can you tell me more about this error, tnx u

C:\CSI\csiread>pip3 install -U dist/csiread*.whl
WARNING: Requirement 'dist/csiread*.whl' looks like a filename, but the file does not exist
Hi, tnx you for 
Can you tell me more about this error
ERROR: csiread*.whl is not a valid wheel filename.
WARNING: You are using pip version 20.2.3; however, version 20.3.3 is available.
You should consider upgrading via the 'c:\users\serrhini\appdata\local\programs\python\python39\python.exe -m pip install --upgrade pip' command.

citysu, 2021-01-13:
`*` is a wildcard character. After run `python3 setup.py sdist bdist_wheel`, you can find a whl file like `csiread-1.3.4-cp36-cp36m-win_amd64.whl` in the dist folder. Replace csiread*.whl with it.

csiread is written in Cython, Cython requires a C compiler to be present on the system. You can refer to [Installing Cython](https://cython.readthedocs.io/en/latest/src/quickstart/install.html) for more details. If you don't want to install a c compiler, just fork the project and push a tag on the last commit. Then wheel files can be found in Github-Actions-Artifacts: csiread_dist

serrhini, 2021-01-16:
tnx you and regards.

closed 2021-02-12 by citysu