citysu/csiread #3: phy_ifft

ENB213-xiaoh, 2021-01-20:
I saw you referenced an equation when measure phy_ifft. The code says it is based on Equation (19-25)(P2373). May I know where it is. its a paper or book?

citysu, 2021-01-20:
IEEE Standard for Information technology—Telecommunications and information exchange between systems Local and metropolitan area networks—Specific requirements - Part 11: Wireless LAN Medium Access Control (MAC) and Physical Layer (PHY) Specifications, in IEEE Std 802.11-2016 (Revision of IEEE Std 802.11-2012), vol., no., pp.1-3534, 14 Dec. 2016, doi: 10.1109/IEEESTD.2016.7786995.

ENB213-xiaoh, 2021-01-20:
btw, the remove_sto in csispofi.py may have issue. The output of polyfit is 2*pi*delta_f*tau instead of tau. Or, maybe im wrong. plz kindly let me know if im wrong.

citysu, 2021-01-20:
Yes, csispotfi.py is unfinished.  Implementation of some functions is incorrect. If you're interested in it. please help me, thanks.

closed 2021-03-10 by ENB213-xiaoh