citysu/csiread #5: can't find and what is Nexmonserver.py and nexmon_server module

serrhini, 2021-05-08:
Thank you for your work, when I try  the nexmoncsi pcap file the following issue arise, thank you for your illumination.
here error what I got  :

C:\CSIKit\csireadnew\csiread-master\examples>python csiserver.py C:\CSIKit\nexmon_csi-feature-python\nexmon_csi-feature-python\utils\python\pcapfiles\output-40.pcap 12 10000
Traceback (most recent call last):
  File "C:\CSIKit\csireadnew\csiread-master\examples\csiserver.py", line 191, in <module>
    nexmon_server(p.csifile, p.number, p.delay)
  File "C:\CSIKit\csireadnew\csiread-master\examples\csiserver.py", line 138, in nexmon_server
    s = socket.socket(socket.PF_PACKET, socket.SOCK_RAW, socket.IPPROTO_IP)
AttributeError: module 'socket' has no attribute 'PF_PACKET'

citysu, 2021-05-08:
- `python csiserver.py ...` is right. `nexmon_server` was the only function in Nexmonserver.py, which has been merged into csiserver.py.
- socket.PF_PACKET is only available on Linux, so csiserver.py(Nexmon) cannot work on Windows

serrhini, 2021-05-13:
Thank you appreciate

closed 2021-05-20 by citysu