citysu/csiread #6: Read CSI data

parthmishra1996, 2021-05-30:
Hello, 

I am having some trouble reading the CSI .dat file while sending and receiving data both the sender and receiver have 2 antennas but it throws an error in the code below:

```python
csifile = "data_session2.dat"
csidata = csiread.Atheros(csifile, nrxnum=2, ntxnum=2, pl_size=10, tones=56)
csidata.read(endian='little')
print(csidata.csi.shape)
```

```python 
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
<ipython-input-123-585d48bc57d5> in <module>
      1 csifile = "data_session2.dat"
      2 csidata = csiread.Atheros(csifile, nrxnum=2, ntxnum=2, pl_size=10, tones=56)
----> 3 csidata.read(endian='little')
      4 print(csidata.csi.shape)

csiread/csiread.pyx in csiread.csiread.Atheros.read()

csiread/csiread.pyx in csiread.csiread.Atheros.read()

csiread/csiread.pyx in csiread.csiread.Atheros.seek()

ValueError: ntxnum=2 is too small!
```

can you help me with this?

citysu, 2021-05-30:
```python
csidata=csiread.Atheros(csifile, nrxnum=3, ntxnum=3, tones=114)
csidata.read('little')
```
After reading, refer to csidata.nr, csidata.nc and csidata.num_tones to correct the parameters.

parthmishra1996, 2021-05-30:
Thanks for your reply!

I did as you suggested.
```python 
csifile = "data_session2.dat"
csidata = csiread.Atheros(csifile, nrxnum=3, ntxnum=3, pl_size=10, tones=56)
csidata.read(endian='little')
print(csidata.csi.shape)
```
```
(1638, 56, 3, 3)
```

These are the output of these parameters, the nc count is 3 so I wonder if this is correct, does this data seem okay? I am trying to use this for some machine learning task. 

[image]

citysu, 2021-05-30:
I'm not sure. Your data seems unstable. csidata.nc and csidata.num_tones(csidata.bandWidth) are changing, I think you missed some operations while collecting data. (You can set nrxnum=np.max(csidata.nr), tones=np.max(csidata.num_tones), etc.)

citysu, 2021-06-08:
I downloaded the data from [Issue: CSI tool not working for Atheros](https://github.com/xieyaxiongfly/Atheros-CSI-Tool-UserSpace-APP/issues/78#issue-912042276), I don't konw how you collected it and what device you used.

I need to know which csitool created the data file. I solved it by `infer_device` in `examples/utils.py`

```python
>>> from utils import infer_device
>>> infer_device('june_3_csidata.dat')
'Atheros'
```

The data was collected by `Atheros CSI Tools`. Let us read it.

```python
>>> import csiread
>>> csidata = csiread.Atheros('june_3_csidata.dat')
>>> csidata.read()
ValueError: ntxnum=2 is too small!
```

I failed to read it with default parameters. The ValueError indicates the transmitter has at least 3 antennas，so I set a larger `ntxnum` and try again.

```python
>>> csidata = csiread.Atheros('june_3_csidata.dat', ntxnum=10)
>>> csidata.read()
3104 packets parsed
```

There are 3104 packets in the csi file (ntxnum > 3 is a rare case, the data may be broken). In addition, the following error may happen.

```python
>>> csidata = csiread.Atheros('june_3_csidata.dat', ntxnum=5)
>>> csidata.read('big')
Segmentation fault (core dumped)
```

This means the csi file was saved on a little-endian computer. you should use `csidata.read('little')` instead of `csidata.read('big')`

```python
>>> import numpy as np
>>> np.unique(csidata.nc)
array([0, 1, 3, 5])
>>> np.unique(csidata.nr)
array([2])
>>> np.unique(csidata.num_tones)
array([ 56, 114])
```

`csidata.nc.max()` is 5, I guess the transmitter has at least 5 antennas. Similarly, the receiver has 2 antennas. so we would use `csiread.Atheros('june_3_csidata.dat', nrxnum=2, ntxnum=5, tones=114)` for less memory next time. As you can see，`csidata.nc` is varying. It is not a constant array as you described above. I suspect that the rate choice is still in automatic mode. Refer to [payload len: 1924](https://github.com/xieyaxiongfly/Atheros-CSI-Tool-UserSpace-APP/issues/22#issuecomment-512192196)

```python
>>> np.unique(csidata.Rate)
array([134, 135, 137, 140, 141, 142, 143])
```

to get stable data. `csidata.Rate` should be a constant array, but it is not.

```python
>>> np.unique(csidata.bandWidth)
array([0, 1])
```

some packets are 20MHz, while othes are 40Mhz. The bandwidth in the transmitter is in automatic mode too.

```python
>>> csidata.csi_len
array([0, 0, 0, ..., 0, 0, 0])
```

`csi_len` is csi data length. Now, it is a zero array, there is no CSI payload saved in the file. Some operations were missed while collecting data. e.g. packet rate. Let's take a look at the timstamp.

```python
>>> timestamp_diff = np.diff(csidata.timestamp)
>>> print((timestamp_diff < 1).sum())
>>> print(csidata.count)
2259
3104
```

The time intervals less than 1 microsecond account for 70%, do we need such a high packet rate?

closed 2021-06-13 by citysu