citysu/csiread #11: nexmon spatial stream index missing

yujianyuanhaha, 2021-07-30:
Greetings,

wonderful job!  I like the neat code style pretty much!

Yet I found it miss spatial stream index  when dealing with 4366c0 (ASUS router) nexmon data, it only output 256 instead of 16*256 CSI data.

Best

Jet

citysu, 2021-07-30:
This can be solved by 

```python
index = np.where((core == 0) & (spatial == 0))[0]
csidata.csi[index]
```

If core and spatial are unknown, `example/csiNexSpatial.py` shows a possible way to restore them, while it is not a perfect solution

yujianyuanhaha, 2021-07-30:
@citysu  thanks for pointing that out.

Say number of package is `N`, the code now output `N * 256`, where their spatial stream index is [0,1,...,15] if you debug into.  What we want is `N*16*256`

citysu, 2021-07-31:
Adjacent packets with the same sequence number(`csidata.seq`) are from the same frame. I've updated `csiNexSpatial.py`, so that you can get the expected shape by

```python
offset = get_frames_index(csidata.seq, 16)
csi = csidata.csi[offset]
sec = csidata.sec[offset]
```

where csi.shape is `frame_count * 16 * 256`.

citysu, 2021-08-02:
@yujianyuanhaha  Now, the example should be OK. You can try it to get the expected shape.

```python
offset = group(csidata.seq, csidata.core, csidata.spatial, 4, 4)
csi = csidata.csi[offset]
sec = csidata.sec[offset]
```

yujianyuanhaha, 2021-08-08:
@citysu  thanks.

How do we import the method `group` in `offset = group(csidata.seq, csidata.core, csidata.spatial, 4, 4)`.  I think it is `from core import group` but it did not work ....

citysu, 2021-08-08:
It has been a method in v1.3.7: `Nexmon.group(c_num=4, s_num=4)`.
None of the examples are in the published package.  they just show you how to implement some functions by yourself

munir01, 2021-09-09:
I tried this. After executing this line:
`offset = csidata.group(4, 4)`
I get this error:
`
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Users/mus1pi/opt/anaconda3/lib/python3.8/site-packages/csiread/core.py", line 597, in group
    return _nex_group(self.seq, self.core, self.spatial, s_num, c_num)
  File "/Users/mus1pi/opt/anaconda3/lib/python3.8/site-packages/csiread/core.py", line 870, in _nex_group
    offset = offset.resahpe(-1, c_num, s_num)
AttributeError: 'numpy.ndarray' object has no attribute 'resahpe'
`
I think, you meant to write `reshape`. I am using v1.3.7. Could you please have a look? Thank you.

citysu, 2021-09-09:
Sorry, it is a typo, It should be `reshape`

munir01, 2021-09-10:
Thanks. I fixed it. Now, it runs.

munir01, 2021-09-10:
Can you please clarify how to use this offset? For example, how to use it to extract the CSI values from antenna 1 or first spatial stream, then antenna 2 or second spatial stream? Also, what is csidata.sec[]? Thank you.

citysu, 2021-09-10:
`core` and `spatial` indicate the spatial stream index. In some nexmon `pcap` files, `core` and `spatial` are always **zero**. I don't know how to solve it. So `Nexmon.group` is currently an experimental feature.

Assume the `core` and `spatial` are known. The shape of `csidata.csi[offset]` and `csidata.sec[offset]` will be (frame, core, spatial, subcarrier) and (frame, core, spatial) respectively. csidata.sec[] is timestamp

munir01, 2021-09-10:
Thanks for the prompt response. It clarifies the situation. When I try `example/csiNexSpatial.py` with my own collected data, I see these figures:
[image]
[image]

My question is: how come the antenna numbers are all the way up to 14? 

This is probably the wrong place to ask: do you know what do they mean by `core` and `spatial` and how to map these to antennas? Thanks so much for your always help!

citysu, 2021-09-11:
The example set `group(c_num=4, s_num=4)`. The ylabel should be spatial stream index(c_num × s_num: 0-15).

Nexmon research is not a part of my work, I know very little about it. I just think `core` is from `rx` and `spatial` is from `tx`

munir01, 2021-09-12:
That helps. Thank you very much!

yujianyuanhaha, 2021-09-21:
@munir01  it is normal, only 14 out of 16 spatial stream appear, on my sides, I usually see 12 out of 16.

yujianyuanhaha, 2021-09-21:
@citysu the codes runs well on some of my data, while broken for some others data,
1. the `csidata` is correct
2. the group into 4x4 is wrong.

 I guess the key is to correctly read the `csidata.core` (usually range 0-3) and `csidata.spatial` (usually range 0-4), once we get them, we can alway easily do the grouping.

Best

Jet

closed 2021-10-31 by citysu