citysu/csiread #12: Nexmon real-time

munir01, 2021-08-31:
Thank you for creating such a very helpful repository. I see that you have support for Nexmon and real-time processing. My question is, how is it supposed to work for real-time processing for Nexmon? I am using an Asus RT-AC86U router to run Nexmon. I see there are two ways to make it happen:

(1) Asus RT-AC86U router can save file in a laptop using Ethernet in real-time. Then we can run csiserver.py by pointing to the file and then csirealtime.py to visualize the content in real-time. 

(2) Similar to the above method, but instead of running the csiserver.py, can we connect csirealtime.py to directly listen to port 5050, where the ASUS RT-AC86U router is publishing? Then, we do not need to create a file and run csiserver.py.

Please let me know your thoughts. Thank you.

citysu, 2021-08-31:
Packets from the router are udp packets. I think `s.bind(('', 5050))` should be OK, nothing else. But I've never tested it.
If it doesn't work, [nexmon_csi:issue195](https://github.com/seemoo-lab/nexmon_csi/issues/195) may help you.

munir01, 2021-08-31:
Thank you for the prompt response. I will try it and let you know.

munir01, 2021-09-05:
Actually, it should be port 5500. I tried and it turns out `s.bind(('', 5500))` does not work the way we expected. I also tried using local port forwarding. That does not work either. However, I was able to make it work using the method (1) I mentioned above. I had to make some changes in the csiserver.py. After that modification, it works. Thanks for the suggestions.

citysu, 2021-09-05:
Thanks for your feedback. You can replace udp socket with raw socket for listening. Try the following code in `csirealtime.py`:

```python
with socket.socket(socket.PF_PACKET, socket.SOCK_RAW, socket.htons(0x3)) as s:
    while True:
        data, address_src = s.recvfrom(4096)
        msg_len = len(data)
        code = self.csidata.pmsg(data[42:])
```

Then `sudo python3 csirealtime.py`, it may work.

closed 2021-10-06 by citysu

zeroby0, 2022-11-13:
For people of the future who come here via search, there are 2 more ways to do it:
- Use ssh port forwarding: https://github.com/seemoo-lab/nexmon_csi/issues/195#issuecomment-800206982
- Use iptables/nftables: https://github.com/nexmonster/relay