citysu/csiread #13: Nexmon real-time parsing issue

munir01, 2021-08-31:
Thanks again for your amazing work. I am experiencing an issue in the Nexmon real-time processing. It could be a bug. I collected the attached .pcap file (csi_aug51_1.pcap) using ASUS RT-AC86U router. I could successfully parse the CSI values using the following code:

``csifile = ".../data/csi_aug25_1.pcap"``
``csidata = csiread.Nexmon(csifile, chip='4366c0', bw=80) ``
``csidata.read()``
``print(csidata.csi.shape)``
``print(csidata.csi[0])``

However, when I try to run it using csirealtime.py and csiserver.py, it does not work. After debugging I found that [line #65 of csirealtime.py](https://github.com/citysu/csiread/blob/1a6150b0a6ca9b417c587c7c7863bb01aee34df8/examples/csirealtime.py#L65) does not evaluate to be true as the[ ``code`` variable at line #53 gets ``N
[csi_aug25_1.zip](https://github.com/citysu/csiread/files/7085767/csi_aug25_1.zip)
one``](https://github.com/citysu/csiread/blob/1a6150b0a6ca9b417c587c7c7863bb01aee34df8/examples/csirealtime.py#L53). Are you using a separate parser for csirealtime.py?

Thank you again for your help.

citysu, 2021-08-31:
Ensure csiread.__version__ >= 1.3.7, and the last line of csirealtime.py is `realtime_plot('nexmon')`

munir01, 2021-08-31:
After upgrading it to version 1.3.7, now it works! Thanks so much!