citysu/csiread #18: Error

CSIhicham, 2022-01-20:
Hello,

I have a problem when I execute this command pip3 install csiread, 
[](url)
[image]

Any help!
Thanks

citysu, 2022-01-21:
Your python version is too low. Please install csiread from source, or install python 3.6+ via anaconda

CSIhicham, 2022-02-22:
> Hello,
> 
> I have a problem when I execute this command pip3 install csiread, [image]
> 
> Any help! Thanks

closed 2022-02-22 by CSIhicham