citysu/csiread #20: # of packets loaded does not match # of packets in the file

RudyGOGOGO, 2022-02-22:
Thanks for your tool!
I simply compared csiread with [CSIKit](https://github.com/Gi-z/CSIKit), yours is much faster.

However, I am facing one weird issue: I used this [Atheros CSI Tool Python RemoteReceive](https://github.com/shuspieler/Athero-CSI-tool-Python-RemoteReceive-Liveview-AmplitudeScaled) to generate a .dat file
In this [script](https://github.com/shuspieler/Athero-CSI-tool-Python-RemoteReceive-Liveview-AmplitudeScaled/blob/master/Server/Atheros_readandplot_in_realtime.py), I set the range with 1000 in this for loop`for i in range()` which should generate a .dat file with 1000 packets.

CSIKit can read the file back with 1000 packets(the shape is [1000, 56, 2, 2]) which seems correct, however, csiread can only retrieve 986 packet(the shape is [986, 56, 2, 2]). I am not sure why there are several packets missing when I use csiread.

P.S. This is the file with CSI data I generated
[with_router_in_realtime.dat.zip](https://github.com/citysu/csiread/files/8114050/with_router_in_realtime.dat.zip)

citysu, 2022-02-22:
This issue was caused by `Atheros CSI Tool Python RemoteReceive`

The format of one atheros packet contains two parts:

| 2 bytes   |  xxx bytes |
|-----------|------------|
| field_len |    payload |

`field_len = len(payload)`. In general, the same device should be used to save both parts, so that the two parts have the same byte order. This is the case that csiread can handle. Differently, your sample data was collected by `Atheros CSI Tool Python RemoteReceive`: `payload` part was first generated on device A(router), then sent to device B(your computer) via udp. `field_len` was calculated on device B. device B is little endian while device A is big endian.

In `CSIKit`, You used

```python
SIZE_STRUCT = struct.Struct("<H").unpack
HEADER_STRUCT_BE = struct.Struct(">QHHBBBBBBBBBBBH").unpack
```

for parsing, so you can retrieve all packets. However, you didn't make the same changes to csiread.

To solve this issue, you can also

- calculate `filed_len` part on device A and send it to device B.

or

- use `filehandle.write(np.uint16(length).newbyteorder())` in function [`save_to_file`](https://github.com/shuspieler/Athero-CSI-tool-Python-RemoteReceive-Liveview-AmplitudeScaled/blob/30ed8566c2e1b54d721e86f0f309084229c94007/Server/Atheros_plotandsave.py#L204) only if you find the two devices have different byte order.

RudyGOGOGO, 2022-02-27:
Sorry for the late reply. I do not get a chance to test until now. Thanks for pointing it out, it is really helpful!!