citysu/csiread #21: Bug when reading large file

maksimio, 2022-02-24:
Hello and thanks for your lib. I have bug with ntxnum in Atheros.read. I am trying to read a large file (150MB) with ntxnum=2, but I am getting the following error:
```
...
File "csiread\_csiread.pyx", line 547, in csiread._csiread.Atheros.seek
File "csiread\_csiread.pyx", line 618, in csiread._csiread.Atheros.seek
ValueError: ntxnum=2 is too small!
```
When i set `ntxnum=5`, and print csidata.csi, it show CSI matrix 2x5, but the elements for 3<=ntxnum<=5 are null. When recording data, the number of antennas Rx is 2 and Tx is 2.
```
...

  [[ 422.-226.j -393.+494.j    0.  +0.j    0.  +0.j    0.  +0.j]
   [-415.-273.j -435. +47.j    0.  +0.j    0.  +0.j    0.  +0.j]]

  [[-399.-311.j  215.+247.j    0.  +0.j    0.  +0.j    0.  +0.j]
   [-382.+439.j -253.-254.j    0.  +0.j    0.  +0.j    0.  +0.j]]

  [[-107.+458.j -200.-413.j    0.  +0.j    0.  +0.j    0.  +0.j]
   [ 135.-452.j -115.+322.j    0.  +0.j    0.  +0.j    0.  +0.j]]]]
```

So why am I getting an error when i set 2<=ntxnum<=4, when real ntxnum is 2?

This is my code:
```python
import csiread

csifile = "./room.dat"
csidata = csiread.Atheros(csifile, nrxnum=2, ntxnum=2, tones=56, if_report=False)
csidata.read(endian='big')
print(csidata.csi)
```
You can access the test large file from the [link](https://drive.google.com/drive/folders/1cKgTZtizBjj3nd_F8H8Tej7fjW7OQwp9)

citysu, 2022-02-24:
The csidata.csi.shape is `[packet_num, tones, nrxnum, ntxnum]`, but for each packet, `[csidata.num_tones, csidata.nr, csidata.nc]` is the real shape. `csidata.csi` is a container.

In your settings, the number of antennas Rx is 2 and Tx is 2. Ideally, nrxnum=2 and ntxnum=2 should work well for csiread. However, in fact, csidata.nr and csidata.nc may be some unexcepted vaules sometimes. It may be caused by some unknown settings. You can plot a figure to check them.

```python
import csiread
import numpy as np
import matplotlib.pyplot as plt

csidata = csiread.Atheros("room.dat", nrxnum=2, ntxnum=5)
csidata.read("big")
print(np.where(csidata.nc == 5))
plt.figure()
plt.plot(csidata.nc)
plt.show()
```

csiread wants to parse all packets instead of skipping the packets with unexcepted `nc`, so you have to set a larger `ntxnum`. After reading, you can filer out outliers by yourself

```python
...
csidata.read("big")
csi = csidata.csi[:, :56, :2, :2]
# or
csi = csidata.csi[csidata.nc == 2]
```

citysu, 2022-02-24:
As a choice, you can modify `_csiread.pyx` to let `csiread.Atheros(..., ntxnum=2)` work.

Replace lines [613-620](https://github.com/citysu/csiread/blob/58c62fc8ff818271f3c9ff84361361d83629df1a/csiread/_csiread.pyx#L613-L620) with:

```python
            c_len = self.buf_csi_len_mem[count]
            if buf[17] > self.nrxnum and c_len != 0:
                fclose(f)
                raise ValueError("nrxnum=%d is too small!\n" % self.nrxnum)
            if buf[18] > self.ntxnum and c_len != 0:
                fclose(f)
                raise ValueError("ntxnum=%d is too small!\n" % self.ntxnum)
```

and replace lines [746-751](https://github.com/citysu/csiread/blob/58c62fc8ff818271f3c9ff84361361d83629df1a/csiread/_csiread.pyx#L746-L751):

```python
            c_len = self.buf_csi_len_mem[count]
            if buf[17] > self.nrxnum and c_len != 0:
                raise ValueError("nrxnum=%d is too small!\n" % self.nrxnum)
            if buf[18] > self.ntxnum and c_len != 0:
                raise ValueError("ntxnum=%d is too small!\n" % self.ntxnum)
```

It seems that `csidata.csi_len` will be `0` when `csidata.nc` is an unexpected value.

maksimio, 2022-02-25:
[citysu](https://github.com/citysu), thank you for your detailed and fast response. Indeed, in a large file there were packets accidentally received from foreign devices. I didn't think to look if there are such packages with ntxnum > 2.

closed 2022-02-25 by maksimio