citysu/csiread #28: How I can 'mute' the annotation like 'connector_log=0x1 1566 0xbb packets parsed'

Ares960826, 2023-01-24:
Thank you for your development. This is a very nice tool and helps many researcher to conduct related works conveniently.

There is an annotation, when the running code econtering `csidata.read()`. Uncomfortably, if I want to parse many `.dat` files, it will output too many annotation in screen.
[image]
At the same time, I try to mute it by the following two methods:
First - 
```python
class HiddenPrints:
    def __enter__(self):
        self._original_stdout = sys.stdout
        sys.stdout = open(os.devnull, 'w')

    def __exit__(self, exc_type, exc_val, exc_tb):
        sys.stdout.close()
        sys.stdout = self._original_stdout
... ... ... ...
with HiddenPrints():
    csidata.read()
```
Second - 
```python
os.system('clear')
```

Please help me out. Thank you again!

Ares960826, 2023-01-24:
These two experiments that I have tried are all not work.

citysu, 2023-01-24:
Take a look at the `docstring` in `core.py`. you can disable the report by setting `if_report=False`.