citysu/csiread #30: All parsed CSI values are 0

munir01, 2023-04-11:
Thank you so much for this amazing project. I was able to extract CSI values using ```csiread``` when the data was collected using Nexmon and ASUS RT-AC86U router. However, when I collect data using a Raspberry Pi 4B using Nexmon, the parsed values seem to be all zeros when I parse using ```csiread```. However, I can see complex numbers using the original Matlab code by [seemoo-lab](https://github.com/seemoo-lab/nexmon_csi/blob/master/utils/matlab/csireader.m).

I am using these instructions:

```
import csiread

csifile = ".../csi_from_rpi_ping.pcap"
csidata = csiread.Nexmon(csifile, chip='bcm43455c0', bw=80) 
csidata.read()
print(csidata.csi[0])

```

My ```csiread``` version is 1.4.0. [Here](https://drive.google.com/file/d/1m6DE_o1ZHbnEUZ12IClftnDd_gXPiXJB/view?usp=sharing) is the .pcap file.

Could you please check why all the parsed CSI values are 0? Thanks in advance!

citysu, 2023-04-12:
`cd example`,

```python
from utils import infer_device, infer_chip_bw
infer_device(csifile)
# 'NexmonPull46'
infer_chip_bw(csifile)
# ('43455c0', 80, 5)
```

so, the code should be:

```python
csifile = "csi_from_rpi_ping.pcap"
csidata = csiread.NexmonPull256(csifile, chip='43455c0', bw=80)
csidata.read()
```

When the argument `chip` is unrecognized, the csi part parsing will be skipped without any warning.

munir01, 2023-04-12:
Thank you for the prompt response! It works now!

But for a [different file](https://drive.google.com/file/d/1ABQqYZYJoVhDC7slYEg-0ia-lA2py_jV/view?usp=sharing) that is captured using the same Raspberry Pi 4B with Nexmon, when I try the following,
```
csifile = "csi_from_rpi_collection2.pcap"

import csiread
from utils import infer_device, infer_chip_bw

infer_device(csifile)
# 'NexmonPull46'
infer_chip_bw(csifile)
# ('43455c0', 80, 5)

csidata = csiread.NexmonPull256(csifile, chip='43455c0', bw=80)
csidata.read()
```
I get this error
[image]

Do you have any suggestions on how to fix it?

Thanks so much again!

citysu, 2023-04-12:
Wireshark shows: The capture file appears to be damaged or corrupt.
(pcap: File has 72155412-byte packet, bigger than maximum of 262144)

munir01, 2023-04-12:
Thank you for checking. This is a limitation of Wireshark only. I can open the file in [Cocoa Packet Analyzer](https://www.tastycocoabytes.com/). Also, I can open the file using original Matlab code by [seemoo-lab](https://github.com/seemoo-lab/nexmon_csi/blob/master/utils/matlab/csireader.m). Regardless of the .pcap file size, will it be possible to parse packet by packet? May be the last packet is faulty, but the rest were just fine?

Also, how can I call the following function in [csiserver.py](https://github.com/citysu/csiread/blob/master/examples/csiserver.py)?
```
csiread.NexmonPull256(csifile, chip='43455c0', bw=80)
```
csiserver.py does not seem to take any arguments to parse using NexmonPull256.

Thank you!

citysu, 2023-04-12:
Yeah, you can parse it partly, if you confirm only the last packet is faulty.

```python
csiread.NexmonPull256(csifile, chip='43455c0', bw=80, bufsize=10000)
````

The capture file is damaged so that `NexmonPull256` cannot get the correct packet count correctly. However, you can tell it by setting bufsize. Set a large value util `csidata.read()` is called successfully.

or

```python
csidata  = csiread.NexmonPull256(None, chip='43455c0', bw=80, bufsize=100)
csidata.seek(csifile, pos=24, num=34)
```
this will parse the first 34 packets, where `pos` should be set by the function in the `examples/csiseek.py`, and the `num <= bufsize` must be True.

munir01, 2023-04-12:
Great! This works in the Terminal! Thanks a lot!

Now, how do I do this in [csiserver](https://github.com/citysu/csiread/blob/master/examples/csiserver.py)? Is there a way to tell it to consider _(a) chip='43455c0', (b) bw=80, and (c) NexmonPull256()_? If I need to hard code these parameters, where do I make that change?

Thanks again!

citysu, 2023-04-14:
csiserver does not need to  parse data completely, it doesn't care the chip and bw. Nexmon and NexmonPull256 can use the same function in csiserver.

munir01, 2023-04-15:
I see. Sure, that makes sense. Thank you!