citysu/csiread #34: how to remove agc (when use get_scaled_csi_sm to get CSI, already remove agc?)

evileleven, 2024-03-20:
get_scaled_csi_sm

citysu, 2024-03-20:
https://github.com/citysu/csiread/blob/c4ec21d5638e2d86887fce557704313bf135e2fd/csiread/_csiread.pyx#L410-L411

evileleven, 2024-03-21:
Thank you so much!

I read a paper (RSSI–CSI Measurement and Variation Mitigation With Commodity Wi-Fi Device), it said that: Pa(dBm) = RSSIa − AGC − C.
[image]

Multiplying the nominal amplitude of CSI by ρ
[image]

get_scaled_csi_sm returns the result as the the result of multiplying the nominal amplitude of CSI by ρ?