citysu/csiread #36: Parsing Nexmon .pcap files

munir01, 2024-07-22:
Thank you for this amazing work! I have a question about parsing CSI and RSSI data using `csiread`. If I know that the CSI data was collected using RPi 4B using [this](https://github.com/nexmonster/nexmon_csi_bin/blob/main/base/5.10.92-v7l%2B.tar.xz) precompiled binary of Nexmon, is there a way to tell if I should use `csiread.Nexmon` or `csiread.NexmonPull46`? Alternatively, given a .pcap file, how can I decide if we should use `csiread.Nexmon` or `csiread.NexmonPull46`? I tried both, but the CSI values are different. Also, there is no `csiread.NexmonPull46.read()` function. So, I created a similar one like `csiread.Nexmon.read()`, which basically calls `super().read()`. Thank you!

citysu, 2024-07-22:
1. When using `NexmonPull46`, if magic number is `0x1111` and rssi (1byte) is not `0x11` and fc (1byte) is not `0x11`, your choice is right. `infer_device` function in `example/utils.py` can also tell you which one should be used.
2. `NexmonPull46` inherits `Nexmon` in `_csiread.pyx`, `NexmonPull46.read()` does exist (but is not wrapped in `core.py`, maybe I forgot it ?).
3. `_autoscale` in `NexmonPull46` is 0 while it is 1 in `Nexmon`. This was introduced in [nexmon_csi pull 256](https://github.com/seemoo-lab/nexmon_csi/pull/256/files#diff-601680361ab8472a9760f2a15319021793745a50edd7e75dbceab543d936df3eR117). You can set `csidata._autoscale = 1` before calling `csidata.read()` if you don't think the change is right.

munir01, 2024-07-23:
Thank you for the prompt response. To follow up,

1. By looking at the values of my magic number (`0x1111`), rssi (192, not `0x11`), and fc (148, not `0x11`), I think I should be able to `NexmonPull46`
2. Yes, I do not see that function in `core.py`. But I added it manually.
3. I did try autoscale, but the values remain the same. Here is the code:
```
csidata = csiread.NexmonPull46(csi_file_path, chip='43455c0', bw=80) 
csidata._autoscale = 1
csidata.read()
csidata.display(0)
```
My concern with RSSI value is that 192 does not seem a reasonable value. But when I used CSIKit library to parse the same file, I get RSSI value -64 for the first packet. I noticed that 192-256=-64. Is there an issue for parsing the RSSI bytes? Thanks again!

citysu, 2024-07-23:
-64 is right, I'll fix it. You can also use `csidata.rssi.astype(np.int8)` to fix the value by yourself.

munir01, 2024-07-24:
`csidata.rssi.astype(np.int8)` works like a charm! Will do. Thanks a lot!