citysu/csiread #38: Intel 5300 .dat file parsing issue

munir01, 2024-10-10:
Thank you for the amazing work! I was trying to parse a .dat file (attached [01_01_01.txt](https://github.com/user-attachments/files/17334829/01_01_01.txt), please rename the extension from .txt to .dat),  which is collected using Intel 5300 chipset with three RX antennas and one TX antenna as a part of [XRF55 dataset](https://github.com/aiotgroup/XRF55-repo). It seems to produce different values with `csiread` library compared to the original Matlab parser with csitool. 

Here are the values from the Matlab parser (attached [XRF55_reader.txt](https://github.com/user-attachments/files/17334868/XRF55_reader.txt), please rename .txt extension to .m)

First antenna, first three packets of first four subcarriers:
subcarrier 1: [7.19770635 8.9468755  7.63842716]
subcarrier 2: [10.92421916 11.56371192 11.23790061]
subcarrier 3: [ 9.26950734 11.71279902 10.65557967]
subcarrier 4: [10.41767222 11.40253732 11.00866919]

Here is the csiread code:
```
csi_file = '../01_01_01.dat'
csidata = csiread.Intel(csi_file, nrxnum=3, ntxnum=1, pl_size=0)
csidata.read()
csi = csidata.get_scaled_csi()
print(csi.shape) # prints (1000, 30, 3, 1)
csi_abs = np.abs(csi)
print(csi_abs[0,0,0,0])
```

It prints 11.090534205814272. The first value should be 7.19770635. Could you please have a look and let me know what could be the reason behind this discrepancy? 

Thank you!

citysu, 2024-10-11:
Thanks for checking it. This is a **bug** introduced in v1.4.0.

It can be fixed by calling `csidata.get_total_rss()` before calling `csidata.get_scaled_csi()` and `csidata.get_scaled_csi_sm()`

munir01, 2024-10-11:
This works! Thanks so much!