citysu/csiread #39: timestamp for data read with csiread.Picoscenes are all zeros?

evileleven, 2024-10-21:
for the same data. 
when use matlab to read it. it works fine.
`uiopen('C:\Users\evelyn\Downloads\wifi.csi',1)
Start parsing PicoScenes CSI file: wifi.csi
141117 PicoScenes frames are decoded in 45.4099 seconds.
wifi`

But with csiread.Picoscenes, it could not give the timestamp. all 'timestamp' are zeros.
`csidata = csiread.Picoscenes(dat_file, {'CSI': [245, 2, 1], 'MPDU': 1522})                
# csidata = csiread.Intel(dat_file, nrxnum=3, ntxnum=1)                                   
csidata.read()                                                                            
csidata.check()                                                                           
print(csidata.raw['CSI']['CSI'].shape)                                                    
                                                                                          
timestamp = csidata.RxSBasic['Timestamp']                                                 
rssi_a = csidata.RxSBasic['RSSI1']                                                        
rssi_b = csidata.RxSBasic['RSSI2']                                                        
csi = csidata.raw['CSI']['CSI']                                                           
csi = csi[:, :, :, 0]                                                                     `

citysu, 2024-10-21:
Use the [official parser](https://gitlab.com/wifisensing/PicoScenes-Python-Toolbox) first. `csiread.Picoscenes` is not recommended because it's too old and lacks maintenance. At the moment it is just a reference for those who want to implement picoscenes parser in python with higher performance than the official implementation.

evileleven, 2024-10-22:
Appreciate your immediate help.
Sometimes, the file is too big making crush even for Matlab parse. When I install package, it give below error even with a total new python3.8 virtual environment. Do you have any idea for it?

> In file included from ./rxs_parsing_core/ModularPicoScenesFrame.hxx:9,
                 from ./picoscenes.cpp:1291:
./rxs_parsing_core/PicoScenesFrameTxParameters.hxx: In member function ‘void PicoScenesFrameTxParameters::applyPreset(const string&)’:
./rxs_parsing_core/PicoScenesFrameTxParameters.hxx:65:49: error: ‘const class std::map<std::__cxx11::basic_string<char>, std::shared_ptr<FrontEndModePreset> >’ has no member named ‘contains’
         if (!FrontEndModePreset::getPresetMap().contains(presetName))
                                                 ^~~~~~~~
In file included from ./rxs_parsing_core/ModularPicoScenesFrame.hxx:11,
                 from ./picoscenes.cpp:1291:
./rxs_parsing_core/CSISegment.hxx: At global scope:
./rxs_parsing_core/CSISegment.hxx:299:36: error: ‘function’ is not a member of ‘std’
     static std::map<uint16_t, std::function<std::shared_ptr<CSI>(const uint8_t *, uint32_t)>> versionedSolutionMap;
                                    ^~~~~~~~
./rxs_parsing_core/CSISegment.hxx:299:36: note: ‘std::function’ is defined in header ‘<functional>’; did you forget to ‘#include <functional>’?
./rxs_parsing_core/CSISegment.hxx:1:1:
+#include <functional>
 //
./rxs_parsing_core/CSISegment.hxx:299:36:
     static std::map<uint16_t, std::function<std::shared_ptr<CSI>(const uint8_t *, uint32_t)>> versionedSolutionMap;
                                    ^~~~~~~~
./rxs_parsing_core/CSISegment.hxx:299:91: error: template argument 2 is invalid
     static std::map<uint16_t, std::function<std::shared_ptr<CSI>(const uint8_t *, uint32_t)>> versionedSolutionMap;
                                                                                           ^
./rxs_parsing_core/CSISegment.hxx:299:91: error: template argument 4 is invalid
./rxs_parsing_core/CSISegment.hxx:299:92: error: expected unqualified-id before ‘>’ token
     static std::map<uint16_t, std::function<std::shared_ptr<CSI>(const uint8_t *, uint32_t)>> versionedSolutionMap;
                                                                                            ^~
./rxs_parsing_core/CSISegment.hxx:301:36: error: ‘function’ is not a member of ‘std’
     static std::map<uint16_t, std::function<std::shared_ptr<CSI>(const uint8_t *, uint32_t)>> initializeSolutionMap() noexcept;
                                    ^~~~~~~~
./rxs_parsing_core/CSISegment.hxx:301:36: note: ‘std::function’ is defined in header ‘<functional>’; did you forget to ‘#include <functional>’?
./rxs_parsing_core/CSISegment.hxx:301:91: error: template argument 2 is invalid
     static std::map<uint16_t, std::function<std::shared_ptr<CSI>(const uint8_t *, uint32_t)>> initializeSolutionMap() noexcept;
                                                                                           ^
./rxs_parsing_core/CSISegment.hxx:301:91: error: template argument 4 is invalid
./rxs_parsing_core/CSISegment.hxx:301:92: error: expected unqualified-id before ‘>’ token
     static std::map<uint16_t, std::function<std::shared_ptr<CSI>(const uint8_t *, uint32_t)>> initializeSolutionMap() noexcept;
                                                                                            ^~
In file included from ./rxs_parsing_core/ModularPicoScenesFrame.hxx:12,
                 from ./picoscenes.cpp:1291:
./rxs_parsing_core/RxSBasicSegment.hxx:69:36: error: ‘function’ is not a member of ‘std’
     static std::map<uint16_t, std::function<RxSBasic(const uint8_t *, uint32_t)>> versionedSolutionMap;
                                    ^~~~~~~~
./rxs_parsing_core/RxSBasicSegment.hxx:69:36: note: ‘std::function’ is defined in header ‘<functional>’; did you forget to ‘#include <functional>’?
./rxs_parsing_core/RxSBasicSegment.hxx:1:1:
+#include <functional>
 //
./rxs_parsing_core/RxSBasicSegment.hxx:69:36:
     static std::map<uint16_t, std::function<RxSBasic(const uint8_t *, uint32_t)>> versionedSolutionMap;
                                    ^~~~~~~~
./rxs_parsing_core/RxSBasicSegment.hxx:69:79: error: template argument 2 is invalid
     static std::map<uint16_t, std::function<RxSBasic(const uint8_t *, uint32_t)>> versionedSolutionMap;
                                                                               ^
./rxs_parsing_core/RxSBasicSegment.hxx:69:79: error: template argument 4 is invalid
./rxs_parsing_core/RxSBasicSegment.hxx:69:80: error: expected unqualified-id before ‘>’ token
     static std::map<uint16_t, std::function<RxSBasic(const uint8_t *, uint32_t)>> versionedSolutionMap;
                                                                                ^~
./rxs_parsing_core/RxSBasicSegment.hxx:71:36: error: ‘function’ is not a member of ‘std’
     static std::map<uint16_t, std::function<RxSBasic(const uint8_t *, uint32_t)>> initializeSolutionMap() noexcept;
                                    ^~~~~~~~
./rxs_parsing_core/RxSBasicSegment.hxx:71:36: note: ‘std::function’ is defined in header ‘<functional>’; did you forget to ‘#include <functional>’?
./rxs_parsing_core/RxSBasicSegment.hxx:71:79: error: template argument 2 is invalid
     static std::map<uint16_t, std::function<RxSBasic(const uint8_t *, uint32_t)>> initializeSolutionMap() noexcept;
                                                                               ^
[... cut ...]
                                                                                                     ^~
./picoscenes.cpp:8776:18: warning: ‘PyObject* __pyx_f_10picoscenes_parse_SignalMatrix(const SignalMatrix<std::complex<float> >*)’ defined but not used [-Wunused-function]
 static PyObject *__pyx_f_10picoscenes_parse_SignalMatrix(SignalMatrix<std::complex<float> >  const *__pyx_v_m) {
                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
error: command '/usr/bin/g++' failed with exit code 1