citysu/csiread #40: Accessing timestamps for NexMon

munir01, 2025-04-01:
Thanks for creating this amazing framework! I am trying to develop a demo that requires access to real-time CSI data and timestamps using NexMon. However, it looks like I can only obtain CSI and RSSI, but not the timestamps. 

I am running a server like this: https://github.com/citysu/csiread/blob/master/examples/csiserver.py

For visualization, I am reading like this:
```
csidata = csiread.NexmonPull46(None, chip='43455c0', bw=80)

   while True:
                # receiving the frames
                frame = conn.recv(BUFFER_SIZE)
                code = csidata.pmsg(frame)  
                print(csidata.csi) # Works well
                
                csidata._autoscale = 1
                csidata.rssi = csidata.rssi.astype(np.int8) 
                print(csidata.rssi) # Works well
                
                print(csidata.sec) # Prints None

```                

How can I access the timestamps? 

Thank you!

citysu, 2025-04-02:
The timestamp is located in pcap packet header. `csiserver.py::nexmon_server()` skips pcap header(16bytes) and enth+ip+udp headers(42bytes). `NexmonPull46.pmsg()` doesn't parse them too, so `csidata.sec`, `csidata.usec`, `csidata.caplen`, etc. are always `None`.

Maybe you can record timestamps by yourself, e.g. `time.time()`.

munir01, 2025-04-02:
Thank you for the detailed explanation!

I agree with your point—capturing headers on the client side makes sense for real-time processing, and that's indeed how I'm handling it during the live demo. I am running `tcpdump -i wlan0 dst port 5500` on the client.

For the demo, I am using server.py to push CSI data to a laptop where I visualize the results. It would be very helpful if server.py could parse the `sec` and `usec` fields, since packet timing is crucial—some algorithms are sensitive to irregular arrival intervals.

Is csiread open source in a way that allows modifications? If so, could you point me to the relevant part of the code where I could add support for parsing those timestamp fields?

Thanks again for your support!

citysu, 2025-04-02:
server side:

```python
    while True:
        if cur >= (lens - 4):
            f.seek(24, os.SEEK_SET)
            cur = 0

        hdr = f.read(16)
        caplen = int.from_bytes(hdr[8:12], byteorder=endian)
        data = hdr + f.read(caplen)
        s.sendto(data, address_des)

        cur += (caplen + 16)
```

client side:

```python
        with socket.socket(socket.AF_INET, socket.SOCK_DGRAM) as s:
            s.bind(self.address_des)
            while self.running:
                data, address_src = s.recvfrom(4096)    # 65507
                msg_len = len(data)

                # pcap header(16bytes)
                hdr = data[:16]
                sec, usec, caplen, wirelen = struct.unpack(endian + "IIII", hdr)

                # enth+ip+udp headers(42bytes), skip

                code = self.csidata.pmsg(data[16 + 42:])
```

munir01, 2025-04-02:
Thanks so much for the prompt response! I tried it and I am getting this error:

 _sec, usec, caplen, wirelen = struct.unpack(endian + "IIII", hdr)
                                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
struct.error: bad char in struct format_

I am using endian="little", which I have observed in NexMon. I have also tried with endian="big", which gives the same error. Do you have any suggestions? 

Thanks in advance!

munir01, 2025-04-02:
This works like a charm! Thanks so much! Really appreciate your help!

munir01, 2025-04-03:
Unfortunately, upon further testing, I realized that it only works with the first packet. The `sec` and `usec` are incorrect from the second packets and all subsequent ones. Do you have any suggestions? Thank you!

citysu, 2025-04-04:
It worked for me. Here is the test code.

suhecheng
***@***.***